HotNoob/PythonProtocolGateway #33: mqtt json no attribute

utdrmac, 2024-05-03:
Just tried setting `json = true` for mqtt

```
May 03 10:39:41 eg4monitor python3[1738]: INFO:classes.transports.transport_base:write data to mqtt transport
May 03 10:39:41 eg4monitor python3[1738]: INFO:classes.transports.transport_base:{'time_year': 24, 'time_month': 5, 'time_date': 3, 'time_hour': 10, 'time_minute': 39, 'time>
May 03 10:39:41 eg4monitor python3[1738]: Traceback (most recent call last):
May 03 10:39:41 eg4monitor python3[1738]:   File "/root/PythonProtocolGateway/protocol_gateway.py", line 200, in run
May 03 10:39:41 eg4monitor python3[1738]:     to_transport.write_data(info)
May 03 10:39:41 eg4monitor python3[1738]:   File "/root/PythonProtocolGateway/classes/transports/mqtt.py", line 173, in write_data
May 03 10:39:41 eg4monitor python3[1738]:     self.client.publish(self.base_topic, json_object, 0, properties=self.__properties)
May 03 10:39:41 eg4monitor python3[1738]:                                                                     ^^^^^^^^^^^^^^^^^
May 03 10:39:41 eg4monitor python3[1738]: AttributeError: 'mqtt' object has no attribute '_mqtt__properties'
May 03 10:39:41 eg4monitor python3[1738]: [2024-05-03 10:39:41,059]  {protocol_gateway.py:205}  ERROR - 'mqtt' object has no attribute '_mqtt__properties'
May 03 10:39:41 eg4monitor python3[1738]: ERROR:invertermodbustomqqt_log:'mqtt' object has no attribute '_mqtt__properties'
May 03 10:39:55 eg4monitor python3[1738]: get registers(0): 12 to 14 (3)
```

utdrmac, 2024-05-03:
I changed it to this:
`            self.client.publish(self.base_topic, json_object, 0, properties=self.mqtt_properties)`
and now it seems to work.

I also noticed that self.mqtt_properties is not sent along when using non-json output.

HotNoob, 2024-05-03:
hmm. to be honest, some residual code from the original project. :P 
since the mqtt server \w home assistant didn't work with the json output.

i'll update the repo for the next release. 

would you say mqtt message expiry is important?  / should be made configurable? 
that's all that  the properties is setting from what i can see.

utdrmac, 2024-05-03:
I don't think that is important, no.  For the record, I'm going MQTT->Telegraf->InfluxDB and telegraf isn't handling all the individual topics well. Telegraf has native json payload decoding, so I'm trying that next. If that doesn't work, then I'm just going to modify the code to write directly to influx, bypassing mqtt altogether.

HotNoob, 2024-05-03:
interesting... 

did some googling. 
influxdb Line Protocol

could definately be implemented as a "transport" in the project.

closed 2024-05-03 by HotNoob

utdrmac, 2024-05-03:
This is troublesome:

```
    "time_year": 24,
    "time_month": 5,
    "time_date": 3,
    "time_hour": 11,
    "time_minute": 17,
    "time_second": 59,
```
I realize this is how the inverter stores it internally. Where would be the best place in the code to put these together as a  timestamp?

HotNoob, 2024-05-03:
not entirely unique to EG4. 

hmmm... in HA i'd just use a script to combine them... 

i think it would be a bit tricky to implement. need to think on this one. 

for the half completed pylon 485 protocol, i have to add a scripting / math language... it could maybe be added when i do that.  
writing to the values wouldn't be easily implemented though. so would be a read-only thing.