HotNoob/PythonProtocolGateway #35: Todo: SRNE inverter protocol v1.7

HotNoob, 2024-05-05:
SRNE is fairly common; protocol is modbus. should be a quick addition.

HotNoob, 2024-05-06:
implemented in v1.1.2
untested.

HotNoob, 2024-05-06:
a lot of weird things in their documentation... such as the signed bits being at the end of the numbers. "sign magnitude numbers" 
 which makes me wonder if they also have high/low bytes backwards. 
 
 so could need a bit of testing to get it going well.

felipecrs, 2024-10-29:
@HotNoob what pdf did you use as base for this one?

HotNoob, 2024-10-29:
https://github.com/HotNoob/PythonProtocolGateway/blob/main/documentation/3rdparty/protocols/SRNE_MODBUS_v3.9.pdf

sorry title is miss-leading. i implemented v3.9
v1.96 was by @gorbyo #45 #47 #48
isn't a pdf in the repo right now. 

i have a v1.7 pdf here though. looks similar to v1.96
[SRNE hybrid solar inverter MODBUS protocol V1 7.pdf](https://github.com/user-attachments/files/17561721/SRNE.hybrid.solar.inverter.MODBUS.protocol.V1.7.pdf)

felipecrs, 2024-10-29:
No no, don't worry. Do it in your own time. But when you do, I can test it for sure.

gorbyo, 2024-10-29:
[SRNE Solar Charge Inverter MODBUS Protocol1.96.pdf](https://github.com/user-attachments/files/17561920/SRNE.Solar.Charge.Inverter.MODBUS.Protocol1.96.pdf)

in attachment modbus for my inverter - ASP/ASF series

HotNoob, 2024-10-30:
#59
implemented v1.7, fully. untested. i left a few registers in that are unique to 1.96 to serve as a marker for me.

in the v1.1.7 branch. 

#60 #61 are required to be done at a later date to "complete" the implementation. for a few spots

alanv72, 2024-11-21:
I have a 240v split phase inverter ASF48100U200-H. Protocol document v1.7 doesn't have Grid Phase B Current register info. v1.96 has it and is working on software 8.03.1.

  - platform: modbus_controller
    modbus_controller_id: srne
    id: ${name}_grid_current_b
    name: "Grid Current B"
    address: 0x0238
    unit_of_measurement: "A"
    register_type: holding
    value_type: U_WORD
    accuracy_decimals: 1
    filters:
      - multiply: 0.1

closed 2025-02-12 by HotNoob