HotNoob/PythonProtocolGateway #39: Proper way to write to registers

utdrmac, 2024-05-09:
What's the correct way to write to a register? I'm not seeing much documentation on this. I just want to write to the registers holding the clock on my inverter to correct the time drift (and probably DST in Nov). I'm thinking this be just a one-shot type script that I can call on a cron job, say every Sunday, to set the clock. I'm not understanding how after I set 'write = true', then where do I specify which regs and what values to write? Thanks.

HotNoob, 2024-05-09:
yes, i should "write" some documentation for writing....

it's still a bit a of a work in progress. currently only single register values are writeable. ( 2 bytes )

so, because of the ambiguous nature of modbus, 
i have a safety feature that requires the holding register to be accurate; i plan on making a per-variable bypass \w manual verification later. 

with that enabled, the writing is done via mqtt. when write is enabled, the script subscribes to
{base topic}/write/{variable name}

with the eg4_v58 protocol, the holding registry map is i think a bit incomplete. 
the verification check is done here:
https://github.com/HotNoob/PythonProtocolGateway/blob/38f1626c0a3342bcfbcb95c5d16d274a2a0c845c/classes/transports/modbus_base.py#L84

so you can bypass that if it doesn't pass, and at the very minimum you'll have to set the value range for the clock. 

oh and the writable column will need to have "W" in it. i left that blank because i was unsure which registers are writable, but i think eg4 just went with the everything in the holding register is writable approach.

HotNoob, 2024-05-09:
oh, i did clean up the eg4_v58 holding register a bit, i think the current version can pass the 90% check. 
https://github.com/HotNoob/PythonProtocolGateway/blob/main/protocols/eg4_v58.holding_registry_map.csv

utdrmac, 2024-07-04:
In reading the code when write is enabled, it looks like there is a subscription for each parameter? Is there some way to filter that, so that instead of 100+ subscriptions, there's only a handful? In my case, I just want to routinely update the time as the EG4 drifts by almost a full minute over the course of a week.

HotNoob, 2024-07-04:
yes, the variable_screen / mask txt files can just remove the variable all together. 

alternatively, editing the protocol csv itself and changing the writable columns to WD or R for readonly.
more info on csv format: https://github.com/HotNoob/PythonProtocolGateway/wiki/Creating-and-Editing-Protocols-%E2%80%90-JSON-%E2%80%90-CSV

when editing the csv, it's best to create a custom protocol so it doesnt get overwritten when updating.
details on making a custom protocol: https://github.com/HotNoob/PythonProtocolGateway/wiki/Protocols#custom--editing-protocols

---
note to self. 
i think in the future, an override csv feature would be a better solution.

aviateur17, 2024-09-24:
@HotNoob, need some help with the writes.

1. Seems I had to change line 127 in modbus_base.py
```
def validate_registry(self, registry_type : Registry_Type = Registry_Type.INPUT) -> float:
```
I think that should be Registry_Type.HOLDING and not Registry_Type.INPUT.  The Holding registers are the writable ones.

2. Then my "score" was 80% which was below the 90% threshold.  What is that "score" based upon?  
 so I had to lower the score_percent on line 88 of modbus_base.py.  Once I can figure out what the score is based upon I can modify my holding register map to increase the score

3. Then I was seeing an exception based on line 309 and 310 of modbus_base.py:
```
File "/home/jlehman/PythonProtocolGateway/classes/transports/modbus_base.py", line 121, in write_data
    self.write_variable(entry, value, Registry_Type.HOLDING)
  File "/home/jlehman/PythonProtocolGateway/classes/transports/modbus_base.py", line 339, in write_variable
    raise ValueError("Invalid value in register. unsafe to write") #i need to figure out a better error handler for theese. 
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: Invalid value in register. unsafe to write
```
```
if not self.protocolSettings.validate_registry_entry(entry, current_value):
            raise ValueError("Invalid value in register. unsafe to write") #i need to figure out a better error handler for theese. 
```
Somehow the validate_registry_entry was returning 0 and not 1 and I have no idea why.  So I went into that function and just put return 1 right after the definition and voila, I was able to write to the register!

So seems like I'm bypassing some of the safeties - the score and then the validate_registry_entry.    If you have any tips on how I can debug this please let me know.

aviateur17, 2024-09-24:
Okay, another question.
Let's say I want to write the time to the inverter. Just testing this out, I'm writing to time_seconds.   So I can see the MQTT entry for time_seconds and then I can see the value written to the inverter upon next query/read.  Then upon successive passes the MQTT entry is ignored unless the value is changed, which is fine and what I'd expect.  However, the issue is when I stop the program and restart the program, upon starting, sees those old, stale, write MQTT entries and then writes them to the inverter upon first pass after startup.   I'd rather have those values not written unless they update while the program is running.  Is there a value I can write to the MQTT entry (maybe here's where the validation comes in) to prevent it from writing upon startup or perhaps we can ignore on startup until the value refreshes in the MQTT server?   Hope all that makes sense.

HotNoob, 2024-09-24:
1. i don't think that would actually have any effect on anything :S
the default parameter is replaced with holding register

https://github.com/HotNoob/PythonProtocolGateway/blob/ed1d92fe72bda5207bccd12ed3375de23485e003/classes/transports/modbus_base.py#L148

although it does look like the code could be cleaned up a bit.

2.

the validation process is just comparing the current holding register values with the .csv / map. checking to see that the values are within the documented ranges. ( values column )

https://github.com/HotNoob/PythonProtocolGateway/blob/main/protocols/eg4/eg4_v58.holding_registry_map.csv

part of the problem is that i don't have ranges for all of the values, so 80% is pretty good. 

your probably safe to adjust this line of code: 
https://github.com/HotNoob/PythonProtocolGateway/blob/ed1d92fe72bda5207bccd12ed3375de23485e003/classes/transports/modbus_base.py#L104

for a long term solution, i can try to find some time to clean up values column a bit, and toss in some vauge interpretations where it's not documented. 

if you could run once with this:
under the modbus portion.

```analyze_protocol = true```

you can stop it after you see "=== START INPUT REGISTER ==="

that should save a copy of your registers into theese files:

input_registry.json
holding_registry.json

then i or you could use them to figure out what the missing value ranges should be. 

3. this would be releated to 2. another "layer" or "safety check"
to write, the minimum is that a valid value range is defined. 

so... tl;dr, need to complete / update the values column in protocols/eg4/eg4_v58.holding_registry_map.csv

---

hmm, the script doesn't store anything after it's restarted. maybe the mqtt broker your using is resending something. 
once the script get's the /write, it should write those changes to the inverter's registers... and then the inverter would decide what todo after that.

HotNoob, 2024-09-24:
i think tonight or tomorrow i'll add a:
``` write = i know what i'm doing ```
sort of thing. 

since there are 2 safety checks.

aviateur17, 2024-09-24:
> Okay, another question. Let's say I want to write the time to the inverter. Just testing this out, I'm writing to time_seconds. So I can see the MQTT entry for time_seconds and then I can see the [...]

I had retain=True on my side when writing to the write/time_seconds topic.  Changing retain=false "fixes" the issue for me.   When restarting the program it no longer writes to that register upon first pass as the topic is now gone.  Just in case others have this same issue.

>if you could run once with this:
under the modbus portion.
analyze_protocol = true
you can stop it after you see "=== START INPUT REGISTER ==="

I will try this. Appreciate your help!

utdrmac, 2025-03-17:
I finally had a chance to sit down, and write a helper script that subscribes to the MQTT topic for my inverter and compare the timestamp produced by the inverter and "now". If there is greater than 30s difference, attempt to write the correct time parts to the topics which PPG is listening on. I was unsuccessful.

```
Mar 17 03:26:16 eg4monitor python3[17352]: INFO:.transport_base[transport.1]:/home/inverter/3162670131/write/time_date 17
Mar 17 03:26:16 eg4monitor python3[17352]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.HOLDING - 13 to 14 (2)
Mar 17 03:26:17 eg4monitor python3[17352]: Exception in thread paho-mqtt-client-:
Mar 17 03:26:17 eg4monitor python3[17352]: Traceback (most recent call last):
Mar 17 03:26:17 eg4monitor python3[17352]:   File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
Mar 17 03:26:17 eg4monitor python3[17352]:     self.run()
Mar 17 03:26:17 eg4monitor python3[17352]:   File "/usr/lib/python3.11/threading.py", line 975, in run
Mar 17 03:26:17 eg4monitor python3[17352]:     self._target(*self._args, **self._kwargs)
Mar 17 03:26:17 eg4monitor python3[17352]:   File "/root/PythonProtocolGateway/lib/python3.11/site-packages/paho/mqtt/client.py", line 4523, in _thread_main
Mar 17 03:26:17 eg4monitor python3[17352]:     self.loop_forever(retry_first_connection=True)
Mar 17 03:26:17 eg4monitor python3[17352]:   File "/root/PythonProtocolGateway/lib/python3.11/site-packages/paho/mqtt/client.py", line 2297, in loop_forever
Mar 17 03:26:17 eg4monitor python3[17352]:     rc = self._loop(timeout)
Mar 17 03:26:17 eg4monitor python3[17352]:          ^^^^^^^^^^^^^^^^^^^
Mar 17 03:26:17 eg4monitor python3[17352]:   File "/root/PythonProtocolGateway/lib/python3.11/site-packages/paho/mqtt/client.py", line 1686, in _loop
Mar 17 03:26:17 eg4monitor python3[17352]:     rc = self.loop_read()
Mar 17 03:26:17 eg4monitor python3[17352]:          ^^^^^^^^^^^^^^^^
Mar 17 03:26:17 eg4monitor python3[17352]:   File "/root/PythonProtocolGateway/lib/python3.11/site-packages/paho/mqtt/client.py", line 2100, in loop_read
Mar 17 03:26:17 eg4monitor python3[17352]:     rc = self._packet_read()
Mar 17 03:26:17 eg4monitor python3[17352]:          ^^^^^^^^^^^^^^^^^^^
Mar 17 03:26:17 eg4monitor python3[17352]:   File "/root/PythonProtocolGateway/lib/python3.11/site-packages/paho/mqtt/client.py", line 3142, in _packet_read
Mar 17 03:26:17 eg4monitor python3[17352]:     rc = self._packet_handle()
Mar 17 03:26:17 eg4monitor python3[17352]:          ^^^^^^^^^^^^^^^^^^^^^
Mar 17 03:26:17 eg4monitor python3[17352]:   File "/root/PythonProtocolGateway/lib/python3.11/site-packages/paho/mqtt/client.py", line 3808, in _packet_handle
Mar 17 03:26:17 eg4monitor python3[17352]:     return self._handle_publish()
Mar 17 03:26:17 eg4monitor python3[17352]:            ^^^^^^^^^^^^^^^^^^^^^^
Mar 17 03:26:17 eg4monitor python3[17352]:   File "/root/PythonProtocolGateway/lib/python3.11/site-packages/paho/mqtt/client.py", line 4145, in _handle_publish
Mar 17 03:26:17 eg4monitor python3[17352]:     self._handle_on_message(message)
Mar 17 03:26:17 eg4monitor python3[17352]:   File "/root/PythonProtocolGateway/lib/python3.11/site-packages/paho/mqtt/client.py", line 4501, in _handle_on_message
Mar 17 03:26:17 eg4monitor python3[17352]:     on_message(self, self._userdata, message)
Mar 17 03:26:17 eg4monitor python3[17352]:   File "/root/PythonProtocolGateway/classes/transports/mqtt_custom.py", line 259, in client_on_message
Mar 17 03:26:17 eg4monitor python3[17352]:     self.on_message(self, entry, msg.payload.decode('utf-8'))
Mar 17 03:26:17 eg4monitor python3[17352]:   File "/root/PythonProtocolGateway/protocol_gateway.py", line 174, in on_message
Mar 17 03:26:17 eg4monitor python3[17352]:     to_transport.write_data({entry.variable_name : data}, transport)
Mar 17 03:26:17 eg4monitor python3[17352]:   File "/root/PythonProtocolGateway/classes/transports/modbus_base.py", line 141, in write_data
Mar 17 03:26:17 eg4monitor python3[17352]:     self.write_variable(entry, value, Registry_Type.HOLDING)
Mar 17 03:26:17 eg4monitor python3[17352]:   File "/root/PythonProtocolGateway/classes/transports/modbus_base.py", line 358, in write_variable
Mar 17 03:26:17 eg4monitor python3[17352]:     raise ValueError(f"Invalid value in register '{current_value}'. Unsafe to write")
Mar 17 03:26:17 eg4monitor python3[17352]: ValueError: Invalid value in register '1041'. Unsafe to write
```

HotNoob, 2025-03-17:
https://github.com/HotNoob/PythonProtocolGateway/blob/ab0416de7ee527b861ec2c27bca2f938b449bed6/classes/transports/modbus_base.py#L359

for the register you are trying to write to ( the protcol csv file ), you need to make sure the value range is defined. 
something like so: 
https://github.com/HotNoob/PythonProtocolGateway/blob/ab0416de7ee527b861ec2c27bca2f938b449bed6/protocols/eg4/eg4_3000ehv_v1.holding_registry_map.csv#L56

0~65535 is ok if your uncertain of the actual range. 

just a "safety" feature.

just assuming the "safety" is correctly working here.

utdrmac, 2025-03-17:
Looks like the ranges are set?

https://github.com/HotNoob/PythonProtocolGateway/blob/ab0416de7ee527b861ec2c27bca2f938b449bed6/protocols/eg4/eg4_v58.holding_registry_map.csv#L12-L17

I did similar to @aviateur17 and just commented out these lines:
https://github.com/HotNoob/PythonProtocolGateway/blob/ab0416de7ee527b861ec2c27bca2f938b449bed6/classes/transports/modbus_base.py#L358-L359

But that results in this error:

```
Mar 17 04:38:18 eg4monitor python3[17458]:   File "/root/PythonProtocolGateway/classes/transports/modbus_base.py", line 141, in write_data
Mar 17 04:38:18 eg4monitor python3[17458]:     self.write_variable(entry, value, Registry_Type.HOLDING)
Mar 17 04:38:18 eg4monitor python3[17458]:   File "/root/PythonProtocolGateway/classes/transports/modbus_base.py", line 405, in write_variable
Mar 17 04:38:18 eg4monitor python3[17458]:     self.write_register(entry.register, ushortValue, registry_type=registry_type)
Mar 17 04:38:18 eg4monitor python3[17458]:   File "/root/PythonProtocolGateway/classes/transports/modbus_rtu.py", line 104, in write_register
Mar 17 04:38:18 eg4monitor python3[17458]:     self.client.write_register(register, value, **kwargs) #function code 0x06 writes to holding register
Mar 17 04:38:18 eg4monitor python3[17458]:     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 17 04:38:18 eg4monitor python3[17458]: TypeError: ModbusClientMixin.write_register() got an unexpected keyword argument 'registry_type'
```

Ref: https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.write_register

HotNoob, 2025-03-17:
That could be a python compat problem.  I'll fix tomorrow morning

utdrmac, 2025-03-17:
I simply removed `registry_type=registry_type` arg from that call and it seems to all work now! I was able to change the hour -1. Seems perhaps there are limitations in modbus on how writing and reading? The script tried changing day, hour, minute, and second back-to-back, but looks like 'second' didn't take affect?  I'm polling every 10s, so maybe I need to change my time keeper monitor to only change 1 registry at a time.

HotNoob, 2025-03-17:
yes modbus does have read/write frequency limitations ect... 
reducing the read rate will increase the write performance. 

on my todo list to add the ability to finetune things.

---

changing mutliple registers at the same time could also hit those modbus limits. roughly max 1 command per second for most devices. so i might need to introduce a write queue to mitigate this problem in the future.

utdrmac, 2025-03-17:
I don't know anything about modbus, but looking at the registry map, is there a way to optimize the writes? For example, I see 'Time_year' at location 12, and 'Time_Month' at location 12.b8. Does that mean year is the first 8 bits of location 12 and month is the 2nd 8 bits of 12?  If that is true, would there be some way to combine two separate writes/updates into a single write?

aviateur17, 2025-03-17:
I was using this program but now I'm using a PLC to interface with my EG4.  I'm reading registers in groups of 40 every second and looping over and over again to get the data from the EG4 inverter.  I read Input registers 0-40 and 80-120 as fast as I can and then every 10 seconds read the rest of the holding and input registers.  I guess the point is that the EG4 can reliably handle polling pretty quickly and reliably which is nice.

Regarding writing the time, I do this as well from the PLC once per day at 2:05AM local time.

You need to bit bang the Month and  year together in one Word (16 bits) and the hours and day into another word and then second and day into a third word.  Then write those three words to the PLC.   I am not up on my python so I don't know how to do this but here is the code I wrote for the PLC.   SHL is shift left and the OR is a bitmask OR.

```
wDateTimeRegs[12] := SHL(UINT_TO_WORD(uMonth),8) OR (UINT_TO_WORD(uYear - 2000) AND 16#00FF); // 16# MMYY
wDateTimeRegs[13] := SHL(UDINT_TO_WORD(Global.clock.oTimeHrs),8) OR (UINT_TO_WORD(uDay) AND 16#00FF); // 16# HHDD
wDatetimeRegs[14] := SHL(UDINT_TO_WORD(Global.clock.oTimeSec),8) OR (UDINT_TO_WORD(Global.clock.oTimeMin) AND 16#00FF); // 16# SSMM
```
So I have an array where each entry in the array is a Word.  Then I take the Month and shift it left 8 bits and bitwise OR that with the year minus 2000 (so 25, currently).  

Then I do the next word of hours and day.   Take the Hour and shift to left 8 bits to be the higher part of the word then bitwise OR that with the Day of the month and store that in the array.

Then the seconds and minutes just like above.

```
mbClient.WriteReg(iReqIdx:=12, iEnable:=bWriteDateTime, iSN:=sn, iUnitId:=1, iStart:=12, iData:=wDateTimeRegs[12]);
mbClient.WriteReg(iReqIdx:=13, iEnable:=bWriteDateTime, iSN:=sn, iUnitId:=1, iStart:=13, iData:=wDateTimeRegs[13]);
mbClient.WriteReg(iReqIdx:=14, iEnable:=bWriteDateTime, iSN:=sn, iUnitId:=1, iStart:=14, iData:=wDateTimeRegs[14]);
```

Then, once per day, I write those three array elements to the EG4 Inverter to Holding Register address 12, 13, and 14 in three, distinct, modbus writes.

If you can convert the same to Python you should be able to do the same.

HotNoob, 2025-03-17:
> I don't know anything about modbus, but looking at the registry map, is there a way to optimize the writes? For example, I see 'Time_year' at location 12, and 'Time_Month' at location 12.b8. Does [...]

yes. a write queue system _could_ reduce the number of write opperations required in this case, if programmed todo so.

utdrmac, 2025-03-17:
@aviateur17 Awesome! Thanks for that info. I might give that a try in my custom module. Also, what's a PLC (assume some sort of hardware device), and which one do you have?

aviateur17, 2025-03-17:
@utdrmac, a PLC is a Programmable Logic Controller, basically a fancy device that runs a program very fast and is reliable.  There are many brands of PLCs but the one I am using is called a Saia Burgess Controls PCD3.M6893.  I work for a company related to that PLC so I didn't have to purchase one.

https://www.saia-pcd.com/en-gb/products/automation-stations/iec-controller-qronox/

One of the programming methods is similar to C programming which I am familiar with.

utdrmac, 2025-03-24:
I'm using this simple command to update the seconds:
```
/ # echo "$(date +%S) + 4" | bc | /usr/bin/mosquitto_pub -t '/home/inverter/3162670131/write/time_second' -s
```
but getting this:

```
Mar 24 20:32:16 eg4monitor python3[26022]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.INPUT - 1 to 37 (37)
Mar 24 20:32:16 eg4monitor python3[26022]: INFO:.transport_base[transport.1]:/home/inverter/3162670131/write/time_second 20
Mar 24 20:32:16 eg4monitor python3[26022]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.HOLDING - 14 to 15 (2)
Mar 24 20:32:17 eg4monitor python3[26022]: INFO:.transport_base[transport.0]:get registers (1): Registry_Type.INPUT - 129 to 130 (2)
Mar 24 20:32:18 eg4monitor python3[26022]: ERROR:classes.protocol_settings:validate_registry_entry fail overall 0 / 59 / 2592
Mar 24 20:32:18 eg4monitor python3[26022]: Exception in thread paho-mqtt-client-:
Mar 24 20:32:18 eg4monitor python3[26022]: Traceback (most recent call last):
Mar 24 20:32:18 eg4monitor python3[26022]:   File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
Mar 24 20:32:18 eg4monitor python3[26022]:     self.run()
Mar 24 20:32:18 eg4monitor python3[26022]:   File "/usr/lib/python3.11/threading.py", line 975, in run
Mar 24 20:32:18 eg4monitor python3[26022]:     self._target(*self._args, **self._kwargs)
Mar 24 20:32:18 eg4monitor python3[26022]:   File "/root/PythonProtocolGateway/lib/python3.11/site-packages/paho/mqtt/client.py", line 4523, in _thread_main
Mar 24 20:32:18 eg4monitor python3[26022]:     self.loop_forever(retry_first_connection=True)
Mar 24 20:32:18 eg4monitor python3[26022]:   File "/root/PythonProtocolGateway/lib/python3.11/site-packages/paho/mqtt/client.py", line 2297, in loop_forever
Mar 24 20:32:18 eg4monitor python3[26022]:     rc = self._loop(timeout)
Mar 24 20:32:18 eg4monitor python3[26022]:          ^^^^^^^^^^^^^^^^^^^
Mar 24 20:32:18 eg4monitor python3[26022]:   File "/root/PythonProtocolGateway/lib/python3.11/site-packages/paho/mqtt/client.py", line 1686, in _loop
Mar 24 20:32:18 eg4monitor python3[26022]:     rc = self.loop_read()
Mar 24 20:32:18 eg4monitor python3[26022]:          ^^^^^^^^^^^^^^^^
Mar 24 20:32:18 eg4monitor python3[26022]:   File "/root/PythonProtocolGateway/lib/python3.11/site-packages/paho/mqtt/client.py", line 2100, in loop_read
Mar 24 20:32:18 eg4monitor python3[26022]:     rc = self._packet_read()
Mar 24 20:32:18 eg4monitor python3[26022]:          ^^^^^^^^^^^^^^^^^^^
Mar 24 20:32:18 eg4monitor python3[26022]:   File "/root/PythonProtocolGateway/lib/python3.11/site-packages/paho/mqtt/client.py", line 3142, in _packet_read
Mar 24 20:32:18 eg4monitor python3[26022]:     rc = self._packet_handle()
Mar 24 20:32:18 eg4monitor python3[26022]:          ^^^^^^^^^^^^^^^^^^^^^
Mar 24 20:32:18 eg4monitor python3[26022]:   File "/root/PythonProtocolGateway/lib/python3.11/site-packages/paho/mqtt/client.py", line 3808, in _packet_handle
Mar 24 20:32:18 eg4monitor python3[26022]:     return self._handle_publish()
Mar 24 20:32:18 eg4monitor python3[26022]:            ^^^^^^^^^^^^^^^^^^^^^^
Mar 24 20:32:18 eg4monitor python3[26022]:   File "/root/PythonProtocolGateway/lib/python3.11/site-packages/paho/mqtt/client.py", line 4145, in _handle_publish
Mar 24 20:32:18 eg4monitor python3[26022]:     self._handle_on_message(message)
Mar 24 20:32:18 eg4monitor python3[26022]:   File "/root/PythonProtocolGateway/lib/python3.11/site-packages/paho/mqtt/client.py", line 4501, in _handle_on_message
Mar 24 20:32:18 eg4monitor python3[26022]:     on_message(self, self._userdata, message)
Mar 24 20:32:18 eg4monitor python3[26022]:   File "/root/PythonProtocolGateway/classes/transports/mqtt_custom.py", line 262, in client_on_message
Mar 24 20:32:18 eg4monitor python3[26022]:     self.on_message(self, entry, msg.payload.decode('utf-8'))
Mar 24 20:32:18 eg4monitor python3[26022]:   File "/root/PythonProtocolGateway/protocol_gateway.py", line 171, in on_message
Mar 24 20:32:18 eg4monitor python3[26022]:     to_transport.write_data({entry.variable_name : data}, transport)
Mar 24 20:32:18 eg4monitor python3[26022]:   File "/root/PythonProtocolGateway/classes/transports/modbus_base.py", line 153, in write_data
Mar 24 20:32:18 eg4monitor python3[26022]:     self.write_variable(entry, value, Registry_Type.HOLDING)
Mar 24 20:32:18 eg4monitor python3[26022]:   File "/root/PythonProtocolGateway/classes/transports/modbus_base.py", line 377, in write_variable
Mar 24 20:32:18 eg4monitor python3[26022]:     raise ValueError(err)
Mar 24 20:32:18 eg4monitor python3[26022]: ValueError: Invalid value in register '2592'. Unsafe to write
```

Not sure where '2592' is coming from. I also notice that if this error happens, the subscriptions to the write topics are lost and you cannot try again. I have to restart the app to regain the topics.

HotNoob, 2025-03-24:
hmm. i have to add 8bit datatype to the validation function. it's currently incomplete.

HotNoob, 2025-03-24:
532d34d99ad9e2674a97ab0300309852b930550a

i think this might do it; i don't have a good setup for testing writes right now. untested.

HotNoob, 2025-03-26:
alrighty. 

68e3fd40be19aa3ddee1d409d90ef8de17d9b192
came up with a way to test modbus writes; simulating a modbus tcp server. should do the trick for testing. 

6f6093a0a147c31127840e46c0f94a837ca84a6f
changed some debugging behaviour and settings. 

https://github.com/HotNoob/PythonProtocolGateway/commit/6f6093a0a147c31127840e46c0f94a837ca84a6f#diff-351d7109db6ad124082df2afa3901da82521abbe797b504ca6b540c7ad8e54a4R16

''' write = true ''' default write behaviour; includes all validations / safties. 
''' write = relaxed ''' skips the initial ( score % ) / bulk validation ( probably what most people will want )
''' write = unsafe ''' skips all write safties. ( dangerous ) 

i'll have to document this properly later. 

anywho, now i can test writing again / better. 

---
note to self; can probably use a register dump to "emulate" devices via modbus_server_sim.py to test things in detail... if ever needed.

HotNoob, 2025-03-26:
> I'm using this simple command to update the seconds:
> 
> ```
> / # echo "$(date +%S) + 4" | bc | /usr/bin/mosquitto_pub -t '/home/inverter/3162670131/write/time_second' -s
> ```

oook. 

2c0a3a51269a78e2afa81d9d9a46362d4dfb8d54

with this. i've tested things via simulated server, and it should be working as intended now.

HotNoob, 2025-07-30:
@ederjc 
sorry, this one should do it:
db29e69f224e09326d0f1fbc162fd708dc46904a

note to self, i need to write some function handlers for codes, before it gets out of hand :P

ederjc, 2025-07-30:
Works perfectly now :))