HotNoob/PythonProtocolGateway #57: Home Assistant add-on

felipecrs, 2024-10-28:
Hi,

Are there any plans to have an add-on for Home Assistant for this project? An add-on would make it very simple for Home Assistant users to get started with it.

I am personally eager to try to use this to integrate my [SRNE HF2430U60-100](https://www.srnesolar.com/productdetail/Off-Grid-Inverters-HF.html) into Home Assistant.

HotNoob, 2024-10-28:
hmmm... it's been suggested before.

looks do-able, but i'd have to learn more about creating a homeassistant add-on. 

i'd probably want to also add a "homeassistant" "transport" in the long-run, to directly add the sensor data to homeassistant and make the mqtt portion optional.

felipecrs, 2024-10-28:
Let me help you with that. Creating an add-on will be easy.

Speaking of homeassistant integration, this page is empty:

https://github.com/HotNoob/PythonProtocolGateway/blob/main/documentation/dashboards/homeassistant.md

I think the app connected to my SRNE inverter, and to the MQTT broker, but I don't know what to do next.

```
Serial Port : /dev/ttyUSB1 =  [0x1a86:0x7523:None:1-1]
[2024-10-28 20:07:00,154]  {protocol_gateway.py:157}  INFO - Connecting to modbus_rtu:transport.modbus...
INFO:invertermodbustomqqt_log:Connecting to modbus_rtu:transport.modbus...
INFO:classes.transports.transport_base:get registers (0): Registry_Type.HOLDING - 24 to 26 (3)
ERROR:classes.transports.transport_base:<bound method ModbusException.__str__ of ModbusIOException('[Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)')>
WARNING:classes.transports.transport_base:Retry(1 - (1)) range(0)
INFO:classes.transports.transport_base:get registers (0): Registry_Type.HOLDING - 24 to 26 (3)
ERROR:classes.transports.transport_base:<bound method ModbusException.__str__ of ModbusIOException('[Input/Output] No Response received from the remote slave/Unable to decode response')>
WARNING:classes.transports.transport_base:Retry(2 - (2)) range(0)
INFO:classes.transports.transport_base:get registers (0): Registry_Type.HOLDING - 24 to 26 (3)
ERROR:classes.transports.transport_base:<bound method ModbusException.__str__ of ModbusIOException('[Input/Output] No Response received from the remote slave/Unable to decode response')>
WARNING:classes.transports.transport_base:Retry(3 - (3)) range(0)
INFO:classes.transports.transport_base:get registers (0): Registry_Type.HOLDING - 24 to 26 (3)
ERROR:classes.transports.transport_base:<bound method ModbusException.__str__ of ModbusIOException('[Input/Output] No Response received from the remote slave/Unable to decode response')>
WARNING:classes.transports.transport_base:Retry(4 - (4)) range(0)
INFO:classes.transports.transport_base:get registers (0): Registry_Type.HOLDING - 24 to 26 (3)
ERROR:classes.transports.transport_base:<bound method ModbusException.__str__ of ModbusIOException('[Input/Output] No Response received from the remote slave/Unable to decode response')>
WARNING:classes.transports.transport_base:Retry(5 - (5)) range(0)
INFO:classes.transports.transport_base:get registers (0): Registry_Type.HOLDING - 24 to 26 (3)
ERROR:classes.transports.transport_base:<bound method ModbusException.__str__ of ModbusIOException('[Input/Output] No Response received from the remote slave/Unable to decode response')>
WARNING:classes.transports.transport_base:Retry(6 - (6)) range(0)
INFO:classes.transports.transport_base:get registers (0): Registry_Type.HOLDING - 24 to 26 (3)
INFO:classes.transports.transport_base:read SN: 
INFO:classes.transports.transport_base:Reading Serial No 1
INFO:classes.transports.transport_base:Reading Serial No 2
INFO:classes.transports.transport_base:Reading Serial No 3
INFO:classes.transports.transport_base:Reading Serial No 4
INFO:classes.transports.transport_base:Reading Serial No 5

[2024-10-28 20:07:42,251]  {protocol_gateway.py:157}  INFO - Connecting to mqtt:transport.mqtt...
INFO:invertermodbustomqqt_log:Connecting to mqtt:transport.mqtt...
INFO:classes.transports.transport_base:mqtt connect
INFO:classes.transports.transport_base:Connected with result code 0
```

No entities appeared in my Home Assistant if it was supposed to.

HotNoob, 2024-10-28:
hmmm... send your config.cfg

looks like it's getting something... after a few tries.

felipecrs, 2024-10-29:
This is the config.cfg:

```
[general]
log_level = DEBUG

[transport.modbus]
log_level = DEBUG

protocol_version = srne_v3.9

address = 1
port = /dev/ttyUSB1
baudrate = 9600

bridge = transport.mqtt

manufacturer = SRNE
model = HF2430U60-100

[transport.mqtt]
log_level = DEBUG
transport = mqtt
host = core-mosquitto
port = 1883
user = mqtt
pass = mypass
```

HotNoob, 2024-10-29:
[transport.modbus]
```
read_interval = 10
```
try setting / increasing the read interval. it could be requesting too often.

[transport.mqtt]
```
discovery_enabled = true
discovery_topic = homeassistant
```
discovery needs to be on for ha to detect anything.

and then, if that doesnt work. 
try 
```
protocol_version = srne_2021_v1.96 
```

felipecrs, 2024-10-29:
Alright. Let me try. By the way, here's the early version of the add-on:

https://github.com/felipecrs/python-protocol-gateway-hass-addon

If you publish your own docker image (from your [Dockerfile](https://github.com/HotNoob/PythonProtocolGateway/blob/main/Dockerfile)), I can also reuse it [like I did for Netdata](https://github.com/felipecrs/netdata-hass-addon).

HotNoob, 2024-10-29:
looks good. 

https://github.com/HotNoob/PythonProtocolGateway-HA-AddOn/
is my attempt; but im banging my head on bugs / documentation discrepancies :P

your approach appears to work better. 

https://github.com/felipecrs/python-protocol-gateway-hass-addon/blob/6de40852dbd36fdc70398edb8e7a4d3529fd3cb1/python-protocol-gateway/rootfs/run.sh#L10
instead of changing the working dir, can use the config parameter to specify the config file

```
protocol_gateway.py --config=config.cfg
```

not sure if changing the dir would have an effect on python imports. 

i rarely use docker :S

felipecrs, 2024-10-29:
Turns out both work, but they provide different entities, as it seems.

`srne_v3.9`:

https://github.com/user-attachments/assets/95009a24-1f6a-4bd7-80e0-778c76eab1f7

`srne_2021_v1.96`:

https://github.com/user-attachments/assets/335d0754-20c5-49fd-a70b-2610fdd1ef32

I don't know yet which one looks better. Both seems to be providing truthful information.

Are you interested to get the _ERROR_ logs?

felipecrs, 2024-10-29:
Looks like `srne_2021_v1.96` is better, as it can read the inverter serial number. I guess I'll stick with it.

This is my "final" `config.cfg`:

```ini
[general]
log_level = WARNING

[transport.modbus]
log_level = WARNING
protocol_version = srne_2021_v1.96
address = 1
port = /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
baudrate = 9600
bridge = transport.mqtt
manufacturer = SRNE
model = HF2430U60-100
read_interval = 10

[transport.mqtt]
log_level = WARNING
transport = mqtt
host = core-mosquitto
port = 1883
user = mqtt
pass = mypass
discovery_enabled = true
discovery_topic = homeassistant
```

And these are the logs (note log_level is WARNING):

[logs.txt](https://github.com/user-attachments/files/17550892/logs.txt)

felipecrs, 2024-10-29:
The add-on is working fine by the way.

felipecrs, 2024-10-29:
Sorry, I meant **my** version of the add-on. I only saw your comment now.

Passing the config file as an argument doesn't seem to be working, regardless of `python3 protocol_gateway.py --config=<config path>` or as in the README `python3 protocol_gateway.py <config path>`.

In both cases, it doesn't seem to pick the config up.

I have made a different approach now, to workaround it: https://github.com/felipecrs/python-protocol-gateway-hass-addon/commit/24db3922020af724b35878b4cbb3fb123bf61496

That's the one which is working.

HotNoob, 2024-10-29:
if you can find another / different srne protocol documentation pdf; it's just a bit of data entry to add it in.

or you can play around with the protocol files, and combine the two, if there is info that you want from both.
just add .custom. to the names so git ignores it / wont be overwritten during updates.

felipecrs, 2024-10-29:
I have no idea how to find the correct pdf for my model. I had sent one email to srne, but they never replied back.

However, I have found some additional pdfs here:

https://github.com/shakthisachintha/SRNE-Hybrid-Inverter-Monitor/tree/master/Resources

Note I don't know yet if I am missing anything. I am just sending this because I found it.

HotNoob, 2024-10-29:
hmm, yeah it usually goes like that :P

v1.7 is a older version of srne_2021_v1.96 , so that's ok
the other is v3.9. 
one is the battery - inverter communcation protocol
and the last is v1.3, which appears to be a older version of v1.7 / 1.96

so i don't think theres anything new there :(

felipecrs, 2024-10-29:
Alright. I guess we can close this issue then. Thanks a lot for everything. I will now play with these data.

If someone else stumbles upon this issue, here is the currently working version of the add-on:

https://github.com/felipecrs/python-protocol-gateway-hass-addon

closed 2024-10-29 by felipecrs

felipecrs, 2024-10-30:
@HotNoob do you think it's normal to get so many of these logs?

```
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc15fb0dd0>>
WARNING:.transport_base[transport.modbus]:Retry(1 - (1)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc154c92b0>>
WARNING:.transport_base[transport.modbus]:Retry(2 - (2)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc15f044a0>>
WARNING:.transport_base[transport.modbus]:Retry(3 - (3)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc15ed6000>>
WARNING:.transport_base[transport.modbus]:Retry(4 - (4)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc15f044a0>>
WARNING:.transport_base[transport.modbus]:Retry(5 - (5)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc15fb1340>>
WARNING:.transport_base[transport.modbus]:Retry(6 - (6)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc15f06000>>
WARNING:.transport_base[transport.modbus]:Retry(7 - (7)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc15ed6000>>
WARNING:.transport_base[transport.modbus]:Retry(8 - (8)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc15da1460>>
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc15f044a0>>
WARNING:.transport_base[transport.modbus]:Retry(8 - (9)) range(5)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc15fb0dd0>>
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc16148b00>>
WARNING:.transport_base[transport.modbus]:Retry(1 - (1)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc16149dc0>>
WARNING:.transport_base[transport.modbus]:Retry(2 - (2)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc16148cb0>>
WARNING:.transport_base[transport.modbus]:Retry(3 - (3)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc16148b00>>
WARNING:.transport_base[transport.modbus]:Retry(4 - (4)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc16149dc0>>
WARNING:.transport_base[transport.modbus]:Retry(5 - (5)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc16148b00>>
WARNING:.transport_base[transport.modbus]:Retry(6 - (6)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc16148cb0>>
WARNING:.transport_base[transport.modbus]:Retry(7 - (7)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc16148b00>>
WARNING:.transport_base[transport.modbus]:Retry(8 - (8)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc16149dc0>>
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc16149d00>>
WARNING:.transport_base[transport.modbus]:Retry(8 - (9)) range(5)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fbc16148b00>>
```

HotNoob, 2024-10-30:
could be caused by too short of an interval if it's random-ish, or those registers ( 3 and 5 ) not being accessible on the device.
some devices can be strict about reading, "write" registers as well. on the todo, to implement a write only mode. 

>normal in a sense. 

a functional for summarizing errors could be something for the future, to keep the logs shorter.

felipecrs, 2024-10-30:
Hm... these errors are consistent, regardless of what I set in `read_interval`. I guess my device don't support it then indeed.

Let me try protocol v1.7 now.

felipecrs, 2024-10-30:
Ok, protocol v1.7 works, but I get a lot more errors:

```
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e48f0>>
WARNING:.transport_base[transport.modbus]:Retry(1 - (1)) range(0)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e73e0>>
WARNING:.transport_base[transport.modbus]:Retry(2 - (2)) range(0)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e48f0>>
WARNING:.transport_base[transport.modbus]:Retry(3 - (3)) range(0)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e73e0>>
WARNING:.transport_base[transport.modbus]:Retry(4 - (4)) range(0)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e48f0>>
WARNING:.transport_base[transport.modbus]:Retry(5 - (5)) range(0)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e73e0>>
WARNING:.transport_base[transport.modbus]:Retry(6 - (6)) range(0)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e48f0>>
WARNING:.transport_base[transport.modbus]:Retry(7 - (7)) range(0)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e73e0>>
WARNING:.transport_base[transport.modbus]:Retry(8 - (8)) range(0)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e4890>>
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e49b0>>
WARNING:.transport_base[transport.modbus]:Retry(7 - (9)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e4890>>
WARNING:.transport_base[transport.modbus]:Retry(8 - (10)) range(3)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e4980>>
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e48f0>>
WARNING:.transport_base[transport.modbus]:Retry(8 - (11)) range(5)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e4920>>
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e49b0>>
WARNING:.transport_base[transport.modbus]:Retry(8 - (12)) range(7)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e73e0>>
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e4980>>
WARNING:.transport_base[transport.modbus]:Retry(6 - (13)) range(11)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e49b0>>
WARNING:.transport_base[transport.modbus]:Retry(7 - (14)) range(11)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e4920>>
WARNING:.transport_base[transport.modbus]:Retry(8 - (15)) range(11)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e4890>>
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e4920>>
WARNING:.transport_base[transport.modbus]:Retry(7 - (16)) range(14)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e48f0>>
WARNING:.transport_base[transport.modbus]:Retry(8 - (17)) range(14)
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e4890>>
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e73e0>>
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e4920>>
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e48f0>>
ERROR:.transport_base[transport.modbus]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7f5b1d2e4890>>
```

Maybe it's just because it has a lot more data. And indeed, it looks like I can view a lot more data.

One initial feedback is that the battery voltages are being calculated in 12v range instead of 24v:

[image]

But the system_voltage is correctly detected:

[image]

Maybe it's possible to multiply these values by system_voltage/12 (multiply by two)?

felipecrs, 2024-10-30:
You can view here that this 12v issue does not happen with the other protocols:

https://github.com/HotNoob/PythonProtocolGateway/issues/57#issuecomment-2443041614

HotNoob, 2024-10-30:
#64 

would probably be the most immediate solution. adding a multipler for the registers.
once it's implemented. 

#28 has some "maths" on the todo list. 

until that's done, can leave it up to home assistant. 

---

if v1.96 is showing the adjusted voltages, it could also be different registers. so when i get to doing a full implementation of v1.96 that could be fixed.

felipecrs, 2024-10-30:
Awesome. Thanks as always.

felipecrs, 2024-10-30:
Further analyzing, there was only one battery-like voltage sensor from 1.96 which is `battery_voltage`, and it was correctly set.

But checking the same sensor in 1.7 also outputs correct number in 24v range.

It looks like only the configuration-like parameters are read as 12v, not the measured ones.

HotNoob, 2024-10-30:
seems about right. 

i'll do #64 tonight. should be quick.

HotNoob, 2024-10-31:
#64 has been implemented in v1.1.7

i'm not entirely confident that it'll perform bug-free, but it works. 

[srne_v1.7.holding_registry_map.override.csv](https://github.com/user-attachments/files/17580512/srne_v1.7.holding_registry_map.override.csv)
this would change those to show as 48v equivalent.

felipecrs, 2025-02-05:
@HotNoob, I've sent a request to transfer the repository over to you. I'm no longer using your application as I have moved on to communication using RS485 with ESPHome instead.