HotNoob/PythonProtocolGateway #65: Todo: Per register timing

HotNoob, 2024-10-30:
per register timing, for better control of how often to read registers. 
this would benefit large register sets; not everything needs to be read at every interval. 

for a format, i'm thinking, *2 or x2, for being every 2nd interval. 

alternatively, perhaphs 30s for every 30 seconds. 

need to think more about this, to implement it in an efficient way.

HotNoob, 2024-10-30:
maybe just a dict \w next interval timestamps, to act as a filter for each read cycle.

HotNoob, 2025-03-17:
33fe4f28fe252f91db579af36281e0ce7ede78dd
v1.1.9

added and debugged. need to document.

to use:

read interval column in protocol or override csv
#x for read_interval from transport config * #
#s for plainold seconds
#ms for miliseconds

read_interval in transport config determins minimum interval.

HotNoob, 2025-03-18:
done. for version v1.1.9

closed 2025-03-18 by HotNoob

utdrmac, 2025-03-23:
This change has introduced some bug/regression because I can't write anymore. In fact, the code isn't subscribing to the /write/ topics.

This is returning empty now (classes/transports/mqtt.py):
`            for entry in from_transport.protocolSettings.get_registry_map(Registry_Type.HOLDING):`

Seems like `def calculate_registry_ranges()` is being called, but the transport has already started processing, thus the transport sees empty ranges. Some sort of race condition? 

I added some debug logging:
```
    def validate_registry(self, registry_type : Registry_Type = Registry_Type.INPUT) -> float:
        score : float = 0
        info = {}

        registry_map : list[registry_map_entry] = self.protocolSettings.get_registry_map(registry_type)
        self._log.info(f"Registry Map: {registry_map}")

        info = self.read_registry(registry_type)
        self._log.info(f"Info map: {info}")
...
```

both of those log lines are producing empty dicts thus validation score is 0 and then no subscribes happen.

HotNoob, 2025-03-23:
shouldn't be effecting the registry map; just the reading. 

https://github.com/HotNoob/PythonProtocolGateway/blob/f5282ccb58664d54c7e73b885cc047fb31745372/classes/transports/mqtt.py#L197-L199

if you haven't updated in a while; the topics were changed to include serial numbers. don't know if you saw that change.

HotNoob, 2025-03-23:
oh...

"            for entry in from_transport.protocolSettings.get_registry_map(Registry_Type.HOLDING): "
is empty? let me test that. could be i screwed up with loading protocolSettings on a per transport level. to get the timing going.

utdrmac, 2025-03-23:
> Lines 197 to 199 in [f5282cc](/HotNoob/PythonProtocolGateway/commit/f5282ccb58664d54c7e73b885cc047fb31745372)
> 
>  topic : str = self.base_topic + "/"+ from_transport.device_identifier + "/write/" + entry.variable_name.lower().replace(" ", "_") 
>  self.__write_topics[topic] = entry 
>  self.client.subscribe(topic) 
> if you haven't updated in a while; the topics were changed to include serial numbers. don't know if you saw that change.

I am up to date with v1.1.19. I added a debug line after the `.subscribe(topic)` to verify the full topic. I no longer see that line in my output, which says something before this function stopped working.

HotNoob, 2025-03-23:
ok. took a bit of effort for my to test it, but the registry_map isn't empty for me. 

i added: 
0763f526d66aa1ed3944d02ff9213427fb05315e

to help test it; to skip the initial validation. 
if you modified that portion to manually skip the protocol validation to enable writting, it could disable writing all together. 
that could also be the problem. 

[image]

HotNoob, 2025-03-23:
sorry, i'm still waking up. found it. 

https://github.com/HotNoob/PythonProtocolGateway/commit/e9253640355010ddd770419275c581122edcdddb#diff-7703307f34b9c27f846c1bb7289ea241ec5f658ab9a9277259642bfa53010736R784

my logic was backwards. so yes the initial registry map was foobar. you had the right spot.

utdrmac, 2025-03-24:
That seems to have fixed it. I can again see the logs I added showing successful subscribe to the write topics. Thanks!