HotNoob/PythonProtocolGateway #68: Received message on invalid discovery topic

felipecrs, 2024-11-13:
I have this in my Home Assistant logs:

```
Logger: homeassistant.components.mqtt.discovery
Source: components/mqtt/discovery.py:395
integration: MQTT (documentation, issues)
First occurred: 10:09:29 AM (1 occurrences)
Last logged: 10:09:29 AM

Received message on illegal discovery topic 'homeassistant/sensor/HN-SR-2404260025-301485-B01//config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic
```

I guess the problem is the `//` part.

This is my config:

```ini
[general]
log_level = INFO

[transport.modbus]
log_level = INFO
protocol_version = srne_v1.7
address = 1
port = /dev/ttyUSB0
baudrate = 9600
read_interval = 10

bridge = transport.mqtt

manufacturer = SRNE
model = HF2430U60-100
serial_number = SR-2404260025-301485-B01
device_name = SRNE

[transport.mqtt]
log_level = INFO
transport = mqtt
host = core-mosquitto
port = 1883
user = mqtt
pass = pass
discovery_enabled = true
discovery_topic = homeassistant

```

HotNoob, 2024-11-13:
good catch. 

simple fix; must have been trying to send a topic for empty registry entries.

25cf59651305a3028973a5001a190230721e1a05
6777e29333c1515ca45149573b916d6f44d447ec - might be a bit redundant.

i do need to add more data-validation / error handling for the csv files.

felipecrs, 2024-11-13:
I can confirm it fixes the issue, thank you.

closed 2024-11-13 by felipecrs