HotNoob/PythonProtocolGateway #72: EG4 MPPT100-48HV Solar Charge Controller Support

frederickjh, 2024-12-29:
Hi!
I was wondering if you know if this will work with the EG4 MPPT100-48HV Solar Charge Controller? It has a RS485 BMS communication port shown in the manual.
Thanks!

HotNoob, 2024-12-29:
that one looks pretty new. 
you can try one of the existing eg4 protocol implementations:

`eg4_v58 = eg4 inverters ( EG4-6000XP ) - confirmed working
eg4_3000ehv_v1 = eg4 inverters ( EG4_3000EHV )`

or acording to the manual:
> "Please contact your distributor to get the Modbus protocol for third-party app development."

if you can get the modbus pdf from them, i can implement the protocol; takes about an hour of data entry todo.

frederickjh, 2024-12-30:
Thanks @HotNoob !

frederickjh, 2024-12-30:
@HotNoob  did a search of the user manual for modbus and only found a place where it talks about how to set the Modbus ID. But I can try and contact them.

HotNoob, 2025-01-28:
@frederickjh i sent eg4 an email and they responded. was not expecting them to respond. 

anyways, they sent me the same pdf that is already in this project -  labeled EG4 3000 EHV-48 - june 2022 v1.0

so setting the protocol_version to eg4_3000ehv_v1 should work.

closed 2025-02-06 by HotNoob

frederickjh, 2025-02-07:
Thanks @HotNoob  for contacting EG4. I had contacted the dealer I bought my unit from and they said I would need to contact EG4 so that was on my to-do list.

I have not yet hooked up my EG4 MPPT100-48HV Solar Charge Controller. I need to build the ground mounts and install the solar panels and wire it all up. However once I get that far I will test with the eg4_3000ehv_v1 protocol and get back to you as to how it works.