HotNoob/PythonProtocolGateway #82: update on EG4 18KPV setup on rPi4

dlarue, 2025-03-26:
Giving your project a go, thank you very much for doing it and sharing. Reason for giving it a go is that our utility changes how much they pay per kWh monthly and each month the rates also vary by hours and days. With batteries I can dump some back to the grid but want to do that based on a downloaded CSV file and will need to change not only the Grid Sell Back Power(kw) but also the Force Discharge Time(s) and Stop Discharge SOC when there's weather in the forecast.

How I setup the rPi4(2GB): note: new rPiOS doesn't like user python packages in system dirs so venv is needed.

SETUP:
get RpiOS Lite, using RPi Imager to set ssh, wifi, user
boot rPi
sudo apt get update & apt get upgrade; reboot
sudo apt install git
git clone https://github.com/HotNoob/PythonProtocolGateway
cd PythonProtocolGateway
sudo apt install python3 pip
python3 -m venv ~/.venv
source ~/.venv/bin/activate
pip install -r requirements
cp config.cfg.example.cfg config.cfg
vi config.cfg ( setup eg4_v58, port /dev/ttyUSB0, 19200 baud and local mqtt server )

source ~/.venv/bin/activate
python3 -u protocol_gateway.py

I connected my USB-RS485 to the 2 spring connectors on the bottom right of the EG4 communications board(left most 1 and 2 connections).
I'm getting data, saw eod, verified by changing it in the web page but could not change it using MQTT command to PythonProtocolGateway.
I've not been able to find the Grid Sell Back Power(kW) protocol message. But today's the first day getting this running.

Cheers.

HotNoob, 2025-03-26:
cool. 

writing is still a feature in it's infancy. 
as it is something that can potentially cause a lot of damage; taking a cautious approach to it with safety mechanisms being implemented.

checkout #39 if you haven't already. 

v1.1.9 has some small changes to the write feature - work-in-progess

HotNoob, 2025-03-26:
not sure if ive documented it yet. 

writes should be sent to a seperate topic:
{base topic}/write/{variable name}

ie:
``` home/ppg/777sim/write/time_minute ```
instead of
``` home/ppg/777sim/time_minute ```

if ppg is getting it, the logs in v1.1.9 will show something like:
``` INFO:.transport_base[transport.mqtt]:MQTT MSG: home/ppg/777sim/write/time_minute 42 ```

can't say about the no password thing... best to at least put some credentials in though :P

dlarue, 2025-03-26:
I've tried:
/usr/bin/mosquitto_pub -h myMqttServerName.local  -t "home/inverter/write/eod" -m 11.0
/usr/bin/mosquitto_pub -h myMqttServerName.local  -t "home/inverter/1234567890/write/eod" -m 11.0

/usr/bin/mosquitto_pub -h myMqttServerName.local -u "myMQTT_USER" -P "" -t "home/inverter/write/eod" -m 11.0
/usr/bin/mosquitto_pub -h myMqttServerName.local -u "myMQTT_USER" -P "" -t "home/inverter/1234567890/write/eod" -m 11.0

Will check for logs, I've been searching console output for feedback on write request.

HotNoob, 2025-03-26:
checkout mqtt-explorer.

https://mqtt-explorer.com/

it's a simple gui mqtt client, makes things easier to visualize / debug the mqtt side of things. 

gl.

dlarue, 2025-03-26:
It's writing!  My mistake was passing 11.0  or 5.2 instead of 11 or 5.
It didn't matter if I used userID and password settings.

I will want to figure out how to send decimal values since if I set grid sell back to 0 my Generator input from micro-inverters is disabled but if I set the grid sell back to 0.1 then the Generator input is enabled and my micro-inverters become operational.

It helped setting ppg to direct std and stderr output to a file and look for the INFO tag as you pointed out.
python3 -u protocol_gateway.py > ../ppg.log 2> >( tee -a ../ppg.log >&2 )
This allowed me to grep for things and see what was going on some.

Speaking of looking at the logs, I do see repeating ERROR msg on data range 94-999 
```
ERROR:.transport_base[transport.0]:<bound method ExceptionResponse.__str__ of <pymodbus.pdu.pdu.ExceptionResponse object at 0x7fa4ad8ad0>>
WARNING:.transport_base[transport.0]:Retry(4 - (4)) range(6)
INFO:.transport_base[transport.0]:get registers (6): Registry_Type.INPUT - 994 to 999 (6)

```

HotNoob, 2025-03-26:
hmm. i don't think i am taking in account for the "units" for writing.
the unit for the register in question would determine the precision of the numbers. modbus itself stores everything as a unsigned short ( whole numbers ). 

---

for the 994 to 999, could be that your inverter doesnt support those registers, or if it's seemingly random, the read interval could be too frequent. you can disable them via variable_screen.txt or for finer control creating a override.csv : https://github.com/HotNoob/PythonProtocolGateway/blob/main/documentation/usage/protocols.md

dlarue, 2025-03-27:
I started a variable_mask.txt file and now that excludes the offending 994-999 message error.

I'm a bit out of my league still with what 'registers' the EG4 18KPV has that we're reading and writing so I need to thread lightly trying to figure out that unit write section would apply.  We are 100% reliant on the inverter/battery/solar so if I put the inverter offline I then need to manually bypass the inverter. Our mains panel is feed from the 18KPV Load output.

I'll look in the protocols/eg4/eg4_v58* files to see if something needs changing to accept additional unit values since there are lots of places I use in the web interface for this level of control granularity.

dlarue, 2025-03-27:
Not that I see there are differences in data types between what the EG4 webpage control uses and what's stored in the inverter, trying to pass a decimal value to a field like EOD( percent now set with 5-95 range) likely isn't needed.
Likewise, MaxBackFlow(=Grid Sell Back Power) which is stored in the inverter as a percentage of some max(10kW?) while the webpage allows decimal settings in kW(ie 0.1 kW = 100W) is a setting of 1%.

Now that I'm becoming more aware of the differences in operational names and data types I'm starting to get somewhere and I'm closing in on starting to write either a bash script or some python code to parse the utility rate schedule(CSV) and adjust the sell back from battery accordingly.

Having a tough time testing right now since during the daylight it seems EG4 servers are getting stretched and often times reading the configuration to populate their web page( or app ) fails to complete. It's more reliable off-peak hours.

dlarue, 2025-03-27:
I believe I have enough known and verified setting to get some control of when and how much exporting to do with excess battery storage. 

Between MaxBackFlow, ForcedDischgSocLimit and the ForcedDischg start/stop Hour, Min variables there should be enough to offset grid connection fees.

This might also force me to get going on setting up Home Assistant on an old x86 table running Linux so there's always an active display to check what's going on.

If ever in San Diego look me up and I'll buy you dinner.

dlarue, 2025-03-31:
I now have to enable/disable Off-Grid Mode( msg 21 ) besides changing the Force-Discharge timer fields. 
Input address 21 is the bit field with Off-grid mode( bit 0 ), so I was wondering if you have any experience writing to the bit fields?  
Can I read the field, convert to HEX, swap the bit, convert to integer and write that integer?

HotNoob, 2025-03-31:
hmmm...  i think i'd have to add in support for writting bitflags.

dlarue, 2025-03-31:
The reading of the bitflags seems to work(except for funcen_espen) since I see funcen_forceddischgen is in the same word and when I query ppg for the value, change it with the webpage, and query then I see the returned value changing from 0 to 1 visa-versa.  I did not try using ppg to write the ForcedDischgEn flag.

Setting a bit like eb7/db7 might be quite handy, binary would be error prone on 16 bit fields because there would be so many bits to count.  And uInt might not be bad but I see that too could be error prone without good documentation. The direct bit manipulation would be easiest on the user and hopefully less prone to errors of changing unintended bits.

Any idea what I did by writing a 1 to FuncEn_EPSEn?  I've not noticed in the EG4 webpage any other fields in the word being out of sorts after it happened.  But I do see that I can no see funcen_epsen change in MQTT Explorer when I make changes using EG4 webpage. Forced Discharge Mode/funcen_forceddischgen  does show changes.

HotNoob, 2025-03-31:
1 would probably have done nothing, as i don't think it would have written anything prior to this update.

---
201a05b4b458c013a5156e385c700aa39a519476
5af8eae6544af73f6144bc6aad8587b96a2a1d36

basic implementation. you can send a "binary string"
left to right. so. 
10000000
would be bit0, 
10000001 bit0 & bit7

---

[image]

in your case ( i have right protocol right? :P )
it looks like everything is split into seperate entries. so with the update, you would just send a single 0 or 1.

HotNoob, 2025-03-31:
oh wait! that's 1bit. not even bit flags :P
uuhhh actually it might not have needed an update..

HotNoob, 2025-03-31:
hmm. did some testing. writing bits function as they should...

not sure why writing for that specific one doesnt work.

dlarue, 2025-03-31:
The thing is, I don't think all of those fields are exposed on the web page so we won't know what to write and not change things. Also, since it's a 16bit field, bit 0 and 7 set would be 1000000100000000 and it's just odd having bit 0 be the left-most bit when hex values are 2s compliment starting at the right(8421).

I was not thinking and had written a 0 and then wrote a 1 and then our inverter rebooted so something didn't go as planned.
  241  /usr/bin/mosquitto_pub -t "home/inverter/xxxxxxxxxx/write/funcen_epsen" -m 0
  242  /usr/bin/mosquitto_pub -t "home/inverter/xxxxxxxxxx/write/funcen_epsen" -m 1

HotNoob, 2025-03-31:
oo. writting rapidly to it could cause some weird behaviour. 

21.b0 vs 21 should be the same. the key is the 1bit data type. 

i've got it on my simulation and it's working as it should for me.
--

little / big endian could be a problem... 
so far i haven't exprienced it in the bitflags yet... so far.

dlarue, 2025-03-31:
I know it looks like it was written quickly but I wrote the 0 and then must have looked on the web page to see it got changed from enable to disable before I then wrote the 1 and that's when the power to the house bounced.

I will try writing the 0 and see what it does on real hardware.

dlarue, 2025-03-31:
I wrote 0 to funen_epsen and it did go to 0( as seen in MQTT Explorer) but my web page didn't change status to disable.
What's more odd is that when I forced my off-grid through the web app, I no longer see it enabling EPS as it did before....
You might be onto something with other settings being in play.

dlarue, 2025-03-31:
I wrote 1 to funen_espen and as before it bounced power but it also looks like EPS is now functioning again.
Need to ask EG4 people about funen_espen/Off-Grid mode to see what's going on behind the scenes. I do not get power bounced when I use the web app to change Off-Grid mode.

dlarue, 2025-03-31:
both funen_espen and FunctionEn1_ubPVGridOffEn stay set at 1 after Off-Grid Mode is disabled in the web app.

dlarue, 2025-04-09:
I'm trying to control AC Couple En and I see it in the input file as unit 77 as follows:
,1bit,77,ACInputType,Bit0,0 or 1,0-Grid 1-Generator for 12KHybrid,,,,,,,,,,,,
,1bit,77.b1,ACCoupleInverterFlow,Bit1,0 or 1,0-no flow 1-show flow,,,,,,,,,,,,
,1bit,77.b2,ACCoupleEn,Bit2,0 or 1,0-Disable 1-Enable,,,,,,,,,,,,

but when I look up the EG4 18kpv modbus protocol it lists unit 77 as
ChgFirstEndHour
ChgFirstEndMinute
which I see setup in the holding file:
,8bit,77,ChgFirstEndHour,hour,0-23,W,Charging priority end time_hour setting,,,,
,8bit,77.b8,ChgFirstEndMinute,min,0-59,W,Charging priority end time_minute setting,,,,

Is there conflicting info there and/or am I missing something?

HotNoob, 2025-04-09:
Holding and Input are just different registers ( adress groups ). Input is
read only.

closed 2025-08-11 by HotNoob