HotNoob/PythonProtocolGateway #83: Inconsistent Registry_Type.HOLDING reads

utdrmac, 2025-03-27:
This seems to be another regression in v1.1.9 where Registry_Type.HOLDING is getting scanned at seemingly random intervals. The main interval is 10s. As the logs below show, sometimes HOLDING is read, and sometimes not. Unsure where to fix this. (Note: Some of the output below is extra that I've added to help diagnose issues) 

```
Mar 27 01:22:14 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.INPUT - 1 to 37 (37)
Mar 27 01:22:15 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (1): Registry_Type.INPUT - 129 to 130 (2)
Mar 27 01:22:16 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.HOLDING - 12 to 14 (3)
Mar 27 01:22:17 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:write data from [transport.0] to mqtt transport
Mar 27 01:22:17 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:{'pv1_voltage': 13.8, 'pv2_voltage': 0.0, 'pv3_voltage': 20.200000000000003, 'soc': 47, 'ppv1': 0.0, 'ppv2': 0.0, 'ppv3': 0.0, 'pcharge': 0.0, 'pdischarge': 341.0, 'pinv': 334.0, 'ptouser': 0.0, 'epv1_day': 0.0, 'epv2_day': 0.0, 'epv3_day': 0.0, 'einv_day': 0.6000000000000001, 'erec_day': 0.0, 'echg_day': 0.0, 'edischg_day': 0.5, 'eeps_day': 0.0, 'etogrid_day': 0.0, 'etouser_day': 0.0, 'peps_l1n': 0.0, 'peps_l2n': 0.0, 'time_year': 25, 'time_month': 3, 'time_date': 27, 'time_hour': 1, 'time_minute': 22, 'time_second': 5}
Mar 27 01:22:17 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:Message 16150 published
Mar 27 01:22:24 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.INPUT - 1 to 37 (37)
Mar 27 01:22:25 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (1): Registry_Type.INPUT - 129 to 130 (2)
Mar 27 01:22:26 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.HOLDING - 12 to 14 (3)
Mar 27 01:22:27 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:write data from [transport.0] to mqtt transport
Mar 27 01:22:27 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:{'pv1_voltage': 13.8, 'pv2_voltage': 0.0, 'pv3_voltage': 20.1, 'soc': 47, 'ppv1': 0.0, 'ppv2': 0.0, 'ppv3': 0.0, 'pcharge': 0.0, 'pdischarge': 438.0, 'pinv': 429.0, 'ptouser': 0.0, 'epv1_day': 0.0, 'epv2_day': 0.0, 'epv3_day': 0.0, 'einv_day': 0.6000000000000001, 'erec_day': 0.0, 'echg_day': 0.0, 'edischg_day': 0.5, 'eeps_day': 0.0, 'etogrid_day': 0.0, 'etouser_day': 0.0, 'peps_l1n': 0.0, 'peps_l2n': 0.0, 'time_year': 25, 'time_month': 3, 'time_date': 27, 'time_hour': 1, 'time_minute': 22, 'time_second': 15}
Mar 27 01:22:27 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:Message 16152 published
Mar 27 01:22:34 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.INPUT - 1 to 37 (37)
Mar 27 01:22:35 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (1): Registry_Type.INPUT - 129 to 130 (2)
Mar 27 01:22:36 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:write data from [transport.0] to mqtt transport
Mar 27 01:22:36 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:{'pv1_voltage': 13.700000000000001, 'pv2_voltage': 0.0, 'pv3_voltage': 20.1, 'soc': 47, 'ppv1': 0.0, 'ppv2': 0.0, 'ppv3': 0.0, 'pcharge': 0.0, 'pdischarge': 489.0, 'pinv': 479.0, 'ptouser': 0.0, 'epv1_day': 0.0, 'epv2_day': 0.0, 'epv3_day': 0.0, 'einv_day': 0.6000000000000001, 'erec_day': 0.0, 'echg_day': 0.0, 'edischg_day': 0.5, 'eeps_day': 0.0, 'etogrid_day': 0.0, 'etouser_day': 0.0, 'peps_l1n': 0.0, 'peps_l2n': 0.0}
Mar 27 01:22:36 eg4monitor python3[27101]: ERROR:.transport_base[transport.1]:Error combining timestamp: 'time_year'
Mar 27 01:22:36 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:Message 16154 published
Mar 27 01:22:45 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.INPUT - 1 to 37 (37)
Mar 27 01:22:46 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (1): Registry_Type.INPUT - 129 to 130 (2)
Mar 27 01:22:47 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.HOLDING - 12 to 14 (3)
Mar 27 01:22:48 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:write data from [transport.0] to mqtt transport
Mar 27 01:22:48 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:{'pv1_voltage': 13.700000000000001, 'pv2_voltage': 0.0, 'pv3_voltage': 20.1, 'soc': 47, 'ppv1': 0.0, 'ppv2': 0.0, 'ppv3': 0.0, 'pcharge': 0.0, 'pdischarge': 451.0, 'pinv': 442.0, 'ptouser': 0.0, 'epv1_day': 0.0, 'epv2_day': 0.0, 'epv3_day': 0.0, 'einv_day': 0.6000000000000001, 'erec_day': 0.0, 'echg_day': 0.0, 'edischg_day': 0.5, 'eeps_day': 0.0, 'etogrid_day': 0.0, 'etouser_day': 0.0, 'peps_l1n': 0.0, 'peps_l2n': 0.0, 'time_year': 25, 'time_month': 3, 'time_date': 27, 'time_hour': 1, 'time_minute': 22, 'time_second': 35}
Mar 27 01:22:48 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:Message 16156 published
Mar 27 01:22:55 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.INPUT - 1 to 37 (37)
Mar 27 01:22:56 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (1): Registry_Type.INPUT - 129 to 130 (2)
Mar 27 01:22:57 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:write data from [transport.0] to mqtt transport
Mar 27 01:22:57 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:{'pv1_voltage': 13.700000000000001, 'pv2_voltage': 0.0, 'pv3_voltage': 20.1, 'soc': 47, 'ppv1': 0.0, 'ppv2': 0.0, 'ppv3': 0.0, 'pcharge': 0.0, 'pdischarge': 472.0, 'pinv': 463.0, 'ptouser': 0.0, 'epv1_day': 0.0, 'epv2_day': 0.0, 'epv3_day': 0.0, 'einv_day': 0.6000000000000001, 'erec_day': 0.0, 'echg_day': 0.0, 'edischg_day': 0.5, 'eeps_day': 0.0, 'etogrid_day': 0.0, 'etouser_day': 0.0, 'peps_l1n': 0.0, 'peps_l2n': 0.0}
Mar 27 01:22:57 eg4monitor python3[27101]: ERROR:.transport_base[transport.1]:Error combining timestamp: 'time_year'
Mar 27 01:22:57 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:Message 16158 published
Mar 27 01:23:05 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.INPUT - 1 to 37 (37)
Mar 27 01:23:06 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (1): Registry_Type.INPUT - 129 to 130 (2)
Mar 27 01:23:07 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.HOLDING - 12 to 14 (3)
Mar 27 01:23:08 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:write data from [transport.0] to mqtt transport
Mar 27 01:23:08 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:{'pv1_voltage': 13.700000000000001, 'pv2_voltage': 0.0, 'pv3_voltage': 20.1, 'soc': 47, 'ppv1': 0.0, 'ppv2': 0.0, 'ppv3': 0.0, 'pcharge': 0.0, 'pdischarge': 438.0, 'pinv': 429.0, 'ptouser': 0.0, 'epv1_day': 0.0, 'epv2_day': 0.0, 'epv3_day': 0.0, 'einv_day': 0.6000000000000001, 'erec_day': 0.0, 'echg_day': 0.0, 'edischg_day': 0.5, 'eeps_day': 0.0, 'etogrid_day': 0.0, 'etouser_day': 0.0, 'peps_l1n': 0.0, 'peps_l2n': 0.0, 'time_year': 25, 'time_month': 3, 'time_date': 27, 'time_hour': 1, 'time_minute': 22, 'time_second': 56}
Mar 27 01:23:08 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:Message 16160 published
Mar 27 01:23:15 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.INPUT - 1 to 37 (37)
Mar 27 01:23:16 eg4monitor python3[27101]: INFO:.transport_base[transport.0]:get registers (1): Registry_Type.INPUT - 129 to 130 (2)
Mar 27 01:23:17 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:write data from [transport.0] to mqtt transport
Mar 27 01:23:17 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:{'pv1_voltage': 13.600000000000001, 'pv2_voltage': 0.0, 'pv3_voltage': 20.0, 'soc': 47, 'ppv1': 0.0, 'ppv2': 0.0, 'ppv3': 0.0, 'pcharge': 0.0, 'pdischarge': 443.0, 'pinv': 434.0, 'ptouser': 0.0, 'epv1_day': 0.0, 'epv2_day': 0.0, 'epv3_day': 0.0, 'einv_day': 0.6000000000000001, 'erec_day': 0.0, 'echg_day': 0.0, 'edischg_day': 0.5, 'eeps_day': 0.0, 'etogrid_day': 0.0, 'etouser_day': 0.0, 'peps_l1n': 0.0, 'peps_l2n': 0.0}
Mar 27 01:23:17 eg4monitor python3[27101]: ERROR:.transport_base[transport.1]:Error combining timestamp: 'time_year'
[... cut ...] rge': 0.0, 'pdischarge': 494.0, 'pinv': 484.0, 'ptouser': 0.0, 'epv1_day': 0.0, 'epv2_day': 0.0, 'epv3_day': 0.0, 'einv_day': 0.6000000000000001, 'erec_day': 0.0, 'echg_day': 0.0, 'edischg_day': 0.5, 'eeps_day': 0.0, 'etogrid_day': 0.0, 'etouser_day': 0.0, 'peps_l1n': 0.0, 'peps_l2n': 0.0, 'time_year': 25, 'time_month': 3, 'time_date': 27, 'time_hour': 1, 'time_minute': 23, 'time_second': 26}
Mar 27 01:23:39 eg4monitor python3[27101]: INFO:.transport_base[transport.1]:Message 16166 published
```

utdrmac, 2025-03-27:
I updated to latest v1.1.9. Restarted. Got the issue right away.

```
Mar 27 01:50:58 eg4monitor python3[28174]: Serial Port : /dev/ttyUSB0 =  [0x403:0x6001:FT8K8WF0:1-1.3]
Mar 27 01:50:59 eg4monitor python3[28174]: [2025-03-27 01:50:59,375]  {protocol_gateway.py:154}  INFO - Connecting to modbus_rtu:transport.0...
Mar 27 01:50:59 eg4monitor python3[28174]: INFO:invertermodbustomqqt_log:Connecting to modbus_rtu:transport.0...
Mar 27 01:50:59 eg4monitor python3[28174]: INFO:.transport_base[transport.0]:Validating Protocol for Writing
Mar 27 01:50:59 eg4monitor python3[28174]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.HOLDING - 12 to 14 (3)
Mar 27 01:51:00 eg4monitor python3[28174]: INFO:.transport_base[transport.0]:validation score: 6 of 6 : 100%
Mar 27 01:51:00 eg4monitor python3[28174]: WARNING:.transport_base[transport.0]:enable write - validation passed
Mar 27 01:51:00 eg4monitor python3[28174]: [2025-03-27 01:51:00,740]  {protocol_gateway.py:154}  INFO - Connecting to mqtt_custom:transport.1...
Mar 27 01:51:00 eg4monitor python3[28174]: INFO:invertermodbustomqqt_log:Connecting to mqtt_custom:transport.1...
Mar 27 01:51:00 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:mqtt connect - /home/inverter/
Mar 27 01:51:00 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:Connected with result code 0
Mar 27 01:51:01 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:Subscribed to /home/inverter/3162670131/write/time_year for WRITE
Mar 27 01:51:01 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:Subscribed to /home/inverter/3162670131/write/time_month for WRITE
Mar 27 01:51:01 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:Subscribed to /home/inverter/3162670131/write/time_date for WRITE
Mar 27 01:51:01 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:Subscribed to /home/inverter/3162670131/write/time_hour for WRITE
Mar 27 01:51:01 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:Subscribed to /home/inverter/3162670131/write/time_minute for WRITE
Mar 27 01:51:01 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:Subscribed to /home/inverter/3162670131/write/time_second for WRITE
Mar 27 01:51:01 eg4monitor python3[28174]: INFO:.transport_base[transport.0]:Register is Empty; transport busy?
Mar 27 01:51:12 eg4monitor python3[28174]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.INPUT - 1 to 37 (37)
Mar 27 01:51:13 eg4monitor python3[28174]: INFO:.transport_base[transport.0]:get registers (1): Registry_Type.INPUT - 129 to 130 (2)
Mar 27 01:51:14 eg4monitor python3[28174]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.HOLDING - 12 to 14 (3)
Mar 27 01:51:15 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:write data from [transport.0] to mqtt transport
Mar 27 01:51:15 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:{'pv1_voltage': 12.8, 'pv2_voltage': 0.0, 'pv3_voltage': 18.1, 'soc': 45, 'ppv1': 0.0, 'ppv2': 0.0, 'ppv3': 0.0, 'pcharge': 0.0, 'pdischarge': 401.0, 'pinv': 393.0, 'ptouser': 0.0, 'epv1_day': 0.0, 'epv2_day': 0.0, 'epv3_day': 0.0, 'einv_day': 0.8, 'erec_day': 0.0, 'echg_day': 0.0, 'edischg_day': 0.7000000000000001, 'eeps_day': 0.0, 'etogrid_day': 0.0, 'etouser_day': 0.0, 'peps_l1n': 0.0, 'peps_l2n': 0.0, 'time_year': 25, 'time_month': 3, 'time_date': 27, 'time_hour': 1, 'time_minute': 51, 'time_second': 2}
Mar 27 01:51:15 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:Message 8 published
Mar 27 01:51:22 eg4monitor python3[28174]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.INPUT - 1 to 37 (37)
Mar 27 01:51:23 eg4monitor python3[28174]: INFO:.transport_base[transport.0]:get registers (1): Registry_Type.INPUT - 129 to 130 (2)
Mar 27 01:51:24 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:write data from [transport.0] to mqtt transport
Mar 27 01:51:24 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:{'pv1_voltage': 12.9, 'pv2_voltage': 0.0, 'pv3_voltage': 18.1, 'soc': 45, 'ppv1': 0.0, 'ppv2': 0.0, 'ppv3': 0.0, 'pcharge': 0.0, 'pdischarge': 407.0, 'pinv': 399.0, 'ptouser': 0.0, 'epv1_day': 0.0, 'epv2_day': 0.0, 'epv3_day': 0.0, 'einv_day': 0.8, 'erec_day': 0.0, 'echg_day': 0.0, 'edischg_day': 0.7000000000000001, 'eeps_day': 0.0, 'etogrid_day': 0.0, 'etouser_day': 0.0, 'peps_l1n': 0.0, 'peps_l2n': 0.0}
Mar 27 01:51:24 eg4monitor python3[28174]: ERROR:.transport_base[transport.1]:Error combining timestamp: 'time_year'
Mar 27 01:51:24 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:Message 10 published
Mar 27 01:51:32 eg4monitor python3[28174]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.INPUT - 1 to 37 (37)
Mar 27 01:51:33 eg4monitor python3[28174]: INFO:.transport_base[transport.0]:get registers (1): Registry_Type.INPUT - 129 to 130 (2)
Mar 27 01:51:34 eg4monitor python3[28174]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.HOLDING - 12 to 14 (3)
Mar 27 01:51:35 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:write data from [transport.0] to mqtt transport
Mar 27 01:51:35 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:{'pv1_voltage': 12.9, 'pv2_voltage': 0.0, 'pv3_voltage': 18.2, 'soc': 45, 'ppv1': 0.0, 'ppv2': 0.0, 'ppv3': 0.0, 'pcharge': 0.0, 'pdischarge': 417.0, 'pinv': 409.0, 'ptouser': 0.0, 'epv1_day': 0.0, 'epv2_day': 0.0, 'epv3_day': 0.0, 'einv_day': 0.8, 'erec_day': 0.0, 'echg_day': 0.0, 'edischg_day': 0.7000000000000001, 'eeps_day': 0.0, 'etogrid_day': 0.0, 'etouser_day': 0.0, 'peps_l1n': 0.0, 'peps_l2n': 0.0, 'time_year': 25, 'time_month': 3, 'time_date': 27, 'time_hour': 1, 'time_minute': 51, 'time_second': 23}
Mar 27 01:51:35 eg4monitor python3[28174]: INFO:.transport_base[transport.1]:Message 12 published
```

HotNoob, 2025-03-27:
hmmm.

if ``` Error combining timestamp: 'time_year' ``` isn't interupting the process or something... 

could just be that the timing isn't normalized, and is calculated when calculate_registry_ranges is called. 

passing transport.last_read_time to calculate_registry_ranges, would be a way to normalize the intervals... but could cause instabilitiy problems. ie, if modbus starts lagging, it won't slow down the requests.

utdrmac, 2025-03-27:
> Error combining timestamp: 'time_year'

This error happens because the holding range, which contains the time components, wasn't polled that interval. It's nothing more than stringing together time_year, time_date, time_month, etc, so it's not the code delaying anything.

This is something that has changed recently because this never happened before, and I have the logs to prove it.

```
Mar 23 00:30:12 eg4monitor python3[19062]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.INPUT - 1 to 37 (37)
Mar 23 00:30:13 eg4monitor python3[19062]: INFO:.transport_base[transport.0]:get registers (1): Registry_Type.INPUT - 129 to 130 (2)
Mar 23 00:30:14 eg4monitor python3[19062]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.HOLDING - 12 to 14 (3)
Mar 23 00:30:15 eg4monitor python3[19062]: INFO:.transport_base[transport.1]:write data from [transport.0] to mqtt transport
Mar 23 00:30:15 eg4monitor python3[19062]: INFO:.transport_base[transport.1]:{'pv1_voltage': 174.9, 'pv2_voltage': 16.3, 'pv3_voltage': 388.8, 'soc': 94, 'ppv1': 19.0, 'ppv2':>
Mar 23 00:30:15 eg4monitor python3[19062]: INFO:.transport_base[transport.1]:Message 59462 published
Mar 23 00:30:22 eg4monitor python3[19062]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.INPUT - 1 to 37 (37)
Mar 23 00:30:23 eg4monitor python3[19062]: INFO:.transport_base[transport.0]:get registers (1): Registry_Type.INPUT - 129 to 130 (2)
Mar 23 00:30:24 eg4monitor python3[19062]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.HOLDING - 12 to 14 (3)
Mar 23 00:30:25 eg4monitor python3[19062]: INFO:.transport_base[transport.1]:write data from [transport.0] to mqtt transport
Mar 23 00:30:25 eg4monitor python3[19062]: INFO:.transport_base[transport.1]:{'pv1_voltage': 187.20000000000002, 'pv2_voltage': 16.400000000000002, 'pv3_voltage': 386.90000000>
Mar 23 00:30:25 eg4monitor python3[19062]: INFO:.transport_base[transport.1]:Message 59464 published
Mar 23 00:30:32 eg4monitor python3[19062]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.INPUT - 1 to 37 (37)
Mar 23 00:30:33 eg4monitor python3[19062]: INFO:.transport_base[transport.0]:get registers (1): Registry_Type.INPUT - 129 to 130 (2)
Mar 23 00:30:34 eg4monitor python3[19062]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.HOLDING - 12 to 14 (3)
Mar 23 00:30:35 eg4monitor python3[19062]: INFO:.transport_base[transport.1]:write data from [transport.0] to mqtt transport
Mar 23 00:30:35 eg4monitor python3[19062]: INFO:.transport_base[transport.1]:{'pv1_voltage': 135.8, 'pv2_voltage': 16.400000000000002, 'pv3_voltage': 384.8, 'soc': 94, 'ppv1':>
Mar 23 00:30:35 eg4monitor python3[19062]: INFO:.transport_base[transport.1]:Message 59466 published
Mar 23 00:30:49 eg4monitor python3[19062]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.INPUT - 1 to 37 (37)
Mar 23 00:30:50 eg4monitor python3[19062]: INFO:.transport_base[transport.0]:get registers (1): Registry_Type.INPUT - 129 to 130 (2)
Mar 23 00:30:51 eg4monitor python3[19062]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.HOLDING - 12 to 14 (3)
Mar 23 00:30:52 eg4monitor python3[19062]: INFO:.transport_base[transport.1]:write data from [transport.0] to mqtt transport
Mar 23 00:30:52 eg4monitor python3[19062]: INFO:.transport_base[transport.1]:{'pv1_voltage': 149.20000000000002, 'pv2_voltage': 16.400000000000002, 'pv3_voltage': 387.0, 'soc'>
Mar 23 00:30:52 eg4monitor python3[19062]: INFO:.transport_base[transport.1]:Message 59468 published
```

Later on Mar 23, I ran into the regression issue I [reported here](https://github.com/HotNoob/PythonProtocolGateway/issues/65#issuecomment-2730714848)

You can see my debug messages, and you can see where I restart the service, after you provided some fixes in #65, then shortly after restart, HOLDING failures:

```
Mar 23 15:29:10 eg4monitor python3[24837]: INFO:classes.protocol_settings:!! ADDED REGISTER time_hour
Mar 23 15:29:10 eg4monitor python3[24837]: INFO:classes.protocol_settings:!! HERE SO FAR time_minute
Mar 23 15:29:10 eg4monitor python3[24837]: INFO:classes.protocol_settings:!! ADDED REGISTER time_minute
Mar 23 15:29:10 eg4monitor python3[24837]: INFO:classes.protocol_settings:!! HERE SO FAR time_second
Mar 23 15:29:10 eg4monitor python3[24837]: INFO:classes.protocol_settings:!! ADDED REGISTER time_second
Mar 23 15:29:10 eg4monitor python3[24837]: INFO:classes.protocol_settings:!! NOT EMPTY
Mar 23 15:29:10 eg4monitor python3[24837]: INFO:classes.protocol_settings:## RANGES [(12, 3)]
Mar 23 15:29:10 eg4monitor python3[24837]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.HOLDING - 12 to 14 (3)
Mar 23 15:29:11 eg4monitor systemd[1]: Stopping protocol_gateway.service - Protocol Gateway...
Mar 23 15:29:11 eg4monitor systemd[1]: protocol_gateway.service: Deactivated successfully.
Mar 23 15:29:11 eg4monitor systemd[1]: Stopped protocol_gateway.service - Protocol Gateway.
Mar 23 15:29:11 eg4monitor systemd[1]: protocol_gateway.service: Consumed 8.139s CPU time.
Mar 23 15:29:11 eg4monitor systemd[1]: Started protocol_gateway.service - Protocol Gateway.
...
Mar 23 15:29:39 eg4monitor python3[24862]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.INPUT - 1 to 37 (37)
Mar 23 15:29:41 eg4monitor python3[24862]: INFO:.transport_base[transport.0]:get registers (1): Registry_Type.INPUT - 129 to 130 (2)
[... cut ...]
Mar 23 15:30:02 eg4monitor python3[24862]: INFO:.transport_base[transport.0]:get registers (0): Registry_Type.HOLDING - 12 to 14 (3)
Mar 23 15:30:03 eg4monitor python3[24862]: INFO:.transport_base[transport.1]:write data from [transport.0] to mqtt transport
Mar 23 15:30:03 eg4monitor python3[24862]: INFO:.transport_base[transport.1]:{'pv1_voltage': 223.5, 'pv2_voltage': 16.5, 'pv3_voltage': 478.8, 'soc': 61, 'ppv1': 21.0, 'ppv2':>
Mar 23 15:30:03 eg4monitor python3[24862]: INFO:.transport_base[transport.1]:Message 10 published
```

HotNoob, 2025-03-27:
a0ccf3c87377adff81a0dbb97d91de98f02a5aa4 - normalize variable reading times

actually, you know what it is... it's the write function. since i'm using calculate_registry_ranges in it as well. 

let me change init=true, to just not touch the timestamp variable all together!
fdc8d08b030280ed3199dea026c084ecf8233977

utdrmac, 2025-03-27:
Looks good! Been running 10m and no missed timestamps. Thanks.