HotNoob/PythonProtocolGateway #89: EG4 18kpv modbus to ip gateway.

BuxtonCalvin, 2025-06-06:
I have an 18k pv inverter and would like to try out the PPG.  I have two requests if possible:

The first is to dockerize the app as running the app on bare metal is a problem for my setup (Ubuntu on a Windows 11 Pro Hyper-V server).  
The second, which is related to the first, is to allow a configuration where I can point to a modbus-to-IP gateway and retrieve inverter/battery data over an ethernet network.
A Hyper-V server does not allow USB connections and though there are ways around this, they are extremely klugey. 

I'd like to buy a converter/gateway off aliexpress and then wire the RS485/RS232 EG4 port to connect to the gateway serial input.  The PPG would then connect as a client to the same gateway via ethernet.  I imagine the gateway mini server would specify a port to listen on.  

Any chance to do something like this?  Thx in advance.

HotNoob, 2025-06-06:
there is a docker file: https://github.com/HotNoob/PythonProtocolGateway/blob/main/Dockerfile
it does also run on windows or linux. 

modbus tcp is suported. 
``` transport = modbus_tcp``` 
instead of
``` transport = modbus_rtu```

HotNoob, 2025-06-06:
https://www.aliexpress.com/item/4000741938697.html?spm=a2g0o.order_list.order_list_main.300.6f5a1802TsOKhv
this is the modbus gateway i personally use with ppg. any should work tho.

BuxtonCalvin, 2025-06-06:
Thank you so much, just what I needed to hear.  I'm going to go with a gateway POE version and it's good to know that this type of setup works.   Is there any chance that you could publish your docker build on docker hub?  I'm asking because the hub allows one to pull images as they are published and as the software is actively worked on, keeping up with bug fixes etc is automatic when a new commit is published to the hub.

HotNoob, 2025-06-06:
https://hub.docker.com/r/hotn00b/pythonprotocolgateway
docker pull hotn00b/pythonprotocolgateway

usage:
```
docker run -v $(pwd)/config.cfg:/app/config.cfg hotn00b/pythonprotocolgateway:latest
```
make sure to pass your config.cfg as /app/config.cfg

---

for person ref for updating later.
```
docker login
docker build . -t hotn00b/pythonprotocolgateway:latest
docker push hotn00b/pythonprotocolgateway:latest
```

BuxtonCalvin, 2025-06-07:
Ok will close for now.  Thanks again.

BuxtonCalvin, 2025-06-07:
> usage:
> 
> ```
> docker run -v $(pwd)/config.cfg:/app/config.cfg hotn00b/pythonprotocolgateway:latest
> ```
> 
> make sure to pass your config.cfg as /app/config.cfg
> 

 /app/config.cfg is showing up as a folder within the image.  So I can't map my local volume config.cfg file to the folder.

HotNoob, 2025-06-07:
[image]

[image]

no idea... 

hmm. maybe something wrong with the run cmd?

BuxtonCalvin, 2025-06-07:
I misnamed the actual local cfg file, so docker created a folder within the image with the mapped name.  Renamed the file and then rebuilt the container and now the cfg is recognized.  Sorry for the trouble.  Will work out the config bugs and report back later.  Thx.

closed 2025-06-07 by BuxtonCalvin